peract/peract #1: When will full implementation code be released?

TongZhangTHU, 2022-09-23:
Hi, thanks for such a wonderful work.
When will full implementation code be released?

MohitShridhar, 2022-09-23:
Hi @TongZhangTHU, I am still cleaning up, but it should be ready in a month or so. Worst case it should be out by the CoRL camera-ready deadline on Nov 15. 

I just pushed my changes to [RLBench](https://github.com/MohitShridhar/RLBench/tree/peract). Still working on YARR and the main repo.

Harimus, 2022-11-21:
Hello, It's past Nov 15 but will the code be uploaded before CoRL?  
We're testing the code based on the available Colab code but it also on  RLBench Pick&Lift where it seems to mistake different object color quite often; is this a limitation of the model? (Looking at the paper result, this seems to look like a problem on some task such as push button)

MohitShridhar, 2022-11-21:
@Harimus I wasted too much time trying to switch to Pytorch Lightning. But I finally managed to get DDP training working. I am writing up a README now. Hopefully, the repo should be ready by this week 🤞

> We're testing the code based on the available Colab code but it also on RLBench Pick&Lift where it seems to mistake different object color quite often

I am not sure about the context, but some things to consider:
- Can you try with more training demos? Like a 100? The rollout visualizations on the website might give you a hint on how many demos are necessary. You shouldn't have any issues with different colors, if you train with sufficient demos. 
- How many iterations did you train the model for? Single-task agents from the paper were trained upto 40K iterations (depends on your batch size etc.)
- Have you seen those colors during training? PerAct doesn't generalize to unseen colors since it's trained from scratch

> problem on some task such as push button

The push button task involves history like "push the red button, and then the blue button". PerAct takes random guesses here since it doesn't really have memory, and is purely a reactive system (as stated in the Limitations).

MohitShridhar, 2022-11-22:
[image]

closed 2022-11-23 by MohitShridhar

Kaixhin, 2022-11-25:
Btw just want to share my appreciation for this. This is fantastic research, your code is neat, your README is extensive, and you're providing datasets and pretrained models. Thank you for the great work :pray:

MohitShridhar, 2022-11-25:
Thanks @Kaixhin! Big fan of your open-source RL work as well!