peract/peract #2: Data of real-world tasks

chenxi-wang, 2022-11-25:
Hi, thanks for your excellent work!
Is there any plan to release the data used in real-world task training?

MohitShridhar, 2022-11-25:
@chenxi-wang here is a [gDrive folder](https://drive.google.com/drive/folders/1plAwdRLgbiEDYyC-Z7kIZpJ2p12CBzZn?usp=share_link) with real-world datasets. If I remember correctly, the Kinect depth images have a different scale to the ones in RLBench. So you have to fix that in the [data-preprocessing code](https://github.com/MohitShridhar/RLBench/blob/peract/rlbench/utils.py#L211).

I don't think these datasets are particularly useful. They are so small, agents trained on them will only work for objects in my lab and my robot setup. Maybe the hand-sanitizer dataset is useful 🤷. The real-world experiments were mostly to show that PerAct _does_ work in the real-world. 

Hope this helps!

chenxi-wang, 2022-11-25:
> @chenxi-wang here is a [gDrive folder](https://drive.google.com/drive/folders/1plAwdRLgbiEDYyC-Z7kIZpJ2p12CBzZn?usp=share_link) with real-world datasets. If I remember correctly, the Kinect depth [...]

Thank you so much!

closed 2022-11-25 by chenxi-wang

baldeeb, 2023-09-28:
For those still interested, here are some [rosbag utilities](https://github.com/baldeeb/rosbag_data_utils) I made to facilitate real robot data translation to training sets. Our lab used this file to translate [bags to RLBench file structure](https://github.com/baldeeb/rosbag_data_utils/blob/main/bag_to_rlbench.py). Hope it helps!