peract/peract #4: V-rep error when generating data

dljzx, 2022-12-01:
Thanks for your good work!
There is a error while I run dataset_generator.py:
Process 0 failed collecting task set_clock_to_time (variation: 2, example: 0). Skipping this task/variation.
The call failed on the V-REP side. Return value: -1

And same error caused in all tasks you have given in your repo of RLbench.
While it seems to be some error in Pyrep, but I can successfully run most of the examples in Pyrep. So how can I solve this problem?

MohitShridhar, 2022-12-01:
@dljzx, sorry I am not sure. This does look like a PyRep issue. Have you seen these threads?
- ARM: https://github.com/stepjam/ARM/issues/2
- RLBench: https://github.com/stepjam/RLBench/search?q=The+call+failed+on+the+V-REP+side.+Return+value%3A+-1&type=issues
- PyRep: https://github.com/stepjam/PyRep/search?q=The+call+failed+on+the+V-REP+side.+Return+value%3A+-1&type=issues

`set_clock_to_time` is something I created, but if all the tasks are failing, then something else is wrong with PyRep or CoppeliaSim. You probably have check with PyRep issues.

dljzx, 2022-12-06:
> @dljzx, sorry I am not sure. This does look like a PyRep issue. Have you seen these threads?
> 
>     * ARM: [Code crashes when trying to run. stepjam/ARM#2](https://github.com/stepjam/ARM/issues/2)
> 
>     * RLBench: https://github.com/stepjam/RLBench/search?q=The+call+failed+on+the+V-REP+side.+Return+value%3A+-1&type=issues
> 
>     * PyRep: https://github.com/stepjam/PyRep/search?q=The+call+failed+on+the+V-REP+side.+Return+value%3A+-1&type=issues
> 
> 
> `set_clock_to_time` is something I created, but if all the tasks are failing, then something else is wrong with PyRep or CoppeliaSim. You probably have check with PyRep issues.

Thanks for your reply. I test the tasks you have given in your version of rlbench. About half of these tasks can successfully generate data, while others still meet the same question mentioned above.
Looking forward to your docker!

MohitShridhar, 2022-12-06:
@dljzx, can you tell me which of these tasks fail:
```
open_drawer
slide_block_to_color_target
sweep_to_dustpan_of_size
meat_off_grill
turn_tap
put_item_in_drawer
close_jar
reach_and_drag
stack_blocks
light_bulb_in
put_money_in_safe
place_wine_at_rack_location
put_groceries_in_cupboard
place_shape_in_shape_sorter
push_buttons
insert_onto_square_peg
stack_cups
place_cups
```

Also, can you post the full stack-trace of the VREP error?

dljzx, 2022-12-07:
> @dljzx, can you tell me which of these tasks fail:
>
> ```
> open_drawer
> slide_block_to_color_target
> sweep_to_dustpan_of_size
> meat_off_grill
> turn_tap
> put_item_in_drawer
> close_jar
> reach_and_drag [...]

After I generate tasks with with the taskid you have listed above, four tasks cannot successfully started which are open_drawer, light_bulb_in, place_wine_at_rack_location and put_item_in_drawer. And also turn_tap stopped in episode 4 with almost same problem that I mentioned above.

OStandardPaths:XDG RUNTIME DIR not set， defaulting totmp/runtime- rootProcess 0 failed collecting task open drawer (variation: l, example: 0). Skipping this task/variation.The call failed on the V-REP side. Return value: -1
Process 0 // Task: turn tap // Variation: 1 // Demo: 0
Process 0 // Task: turn tap // Variation: 0 // Demo: 0
[1206 14:15:35.295553 140313610352448 task enviromentpy:153] Bad demo. The call failed on the v-RE side. Return value: -1 Atempts left; 9Process 0 // Task: turn tap // Variation: 0 // Demo: 0
[1206 14:15:36275046 140313610352448 task enviromentpy:1531 Bad demo, The call failed on the v-REP side. Return value: -1 Atempts left:9Process 0 // Task: turn tap // Variation: 0 // Demo: 0
[1206 14:15:37574196 140313610352448 task enviroment.py:153] Bad demo, The call failed on the V-REP side. Return value: -1 Atempts left; 9Process 0 // Task: turn tap // Variation: 0 // Demo: 0Process 0 // Task: turn tap // Variation: 1 // Demo: 0

MohitShridhar, 2022-12-08:
@dljzx, it looks like the motion-planner is failing to reach waypoints in the demonstration. It could be due to PyRep and CoppeliaSim mismatch. Make sure they [match](https://github.com/stepjam/ARM/issues/2).

Also, does `Demo: 0` ever increase to `Demo: 1`? `Bad demo`s are expected because some target poses are unreachable. It should discard that scene and move on. But if `Demo: 0` never increases, then it's an issue.

dljzx, 2022-12-08:
> @dljzx, it looks like the motion-planner is failing to reach waypoints in the demonstration. It could be due to PyRep and CoppeliaSim mismatch. Make sure they [...]

PyRep and CoppeliaSim I install by following the pyrep github page, CoppeliaSim 4.1.0 with PyRep 4.1.0.3. While I ran the code within docker container that I created. Maybe I think trouble comes from the docker image . It is glad that if you can share your docker on hub or DockerFile as well