peract/peract #6: Confusion regarding demo augmentation

zhouxian, 2022-12-15:
Thanks for releasing the code! It's really helpful.
I have one confusion about the demo augmentation process.

In the inner loop of `fill_replay()`, line 256 of `launch_utils.py`, you call `_add_keypoints_to_replay()` every `demo_augmentation_every_n steps`. For example, if `i = 10` and if the `episode_keypoints` is [33, 43, 55], then in  `_add_keypoints_to_replay()` the following transitions will be added:

obs of frame 10, target gripper pose from frame 33,
obs of frame 33, target gripper pose from frame 43,
obs of frame 43, target gripper pose from frame 55,

The latter 2 are not dependent on `i` in the loop. Therefore, for `i = 20`, the latter 2 transitions will be added to the replay again. Is this what's happening? Or is my understanding not correct?

Thanks!

MohitShridhar, 2022-12-16:
@zhouxian this is a design choice from [ARM (by James et al.)](https://github.com/stepjam/ARM/blob/main/arm/c2farm/launch_utils.py#L161). I am guessing the keyframes get added several times because they indicate important "phase transitions" between bottlenecks, and having several copies makes them more likely to be sampled. But maybe it's a leftover artifact from something else; you can ask [here](https://github.com/stepjam/ARM/issues)

zhouxian, 2022-12-16:
I see. That makes sense to me. Thanks!

closed 2022-12-16 by zhouxian