peract/peract #7: Question about languag embedding

mkt1412, 2022-12-18:
Hi,

Thanks for sharing the work! I have one simple question related to language embedding and data fusion. For CLIP language encoder, it receives a language instruction of variable length and generates a [77, 512] dim language embedding. Most of the tokens are padded. However, when fusing language embeddings and visual representations, no masks are actually used for padded tokens. Will this affect the overall performance?

Thanks!

MohitShridhar, 2022-12-19:
@mkt1412, yes most of the language input is just padding. But this shouldn't cause any issues as long as it's used consistently between train and test time. In vision-language Transformers (like LXMERT, VilBERT), maybe this matters because the visual and textual input could be of different lengths, but in PerAct the language sequence and voxel patch sequence are always of fixed length, so it doesn't matter much.

mkt1412, 2022-12-20:
@MohitShridhar Thanks Mohit!

closed 2022-12-20 by mkt1412