peract/peract #8: Change colors and textures for perturbation tests

geyuying, 2023-01-04:
Thanks for your excellent work!

For the perturbation tests in the paper, you test the model  on ''open drawer'', where handles have randomized textures and colors.  How do you change the textures and colors of the drawer when generating the data useing RLBench？

MohitShridhar, 2023-01-05:
@geyuying, I manually edited the scene files in CoppeliaSim. It's painfully slow because you have to edit one part at a time with the GUI editor. Here is a [zip file](https://github.com/peract/peract/files/10349322/analysis.zip) of `ttm` and `task.py` files from the perturbation tests in the paper. You have to put the `ttm` files inside `RLBench/rlbench/task_ttms` and python files inside `RLBench/rlbench/tasks`. 

Also take a look at [VLMBench](https://github.com/eric-ai-lab/VLMbench). They probably have a better way of randomizing colors and textures.

geyuying, 2023-01-05:
Thanks a lot for your help!

closed 2023-01-05 by geyuying