peract/peract #9: Action Shape Exception during Evaluation

k2so3, 2023-01-05:
Hi Mohit,

Thanks for sharing your code! It's been a really helpful reference. I'm trying to evaluate your pre-trained PerAct model locally with the `open_drawer` task, but I'm running into the following error (which I only found on the tensorboard log and not via the console output) 

`Success: False | Errors - IK : 0, ConfigPath : 0, InvalidAction : 6 Last Exception: Expected the action shape to be: (1,), but was shape: (2,)`

The error happens after the arm moves to the correct drawer, right before the gripper closes to grab the drawer's handle. 

I did some digging and I think this error is coming from either `rlbench::action_modes::gripper_action_modes::assert_action_shape` or `rlbench::action_modes::arm_action_modes::assert_action_shape`. I have `git clone`-ed your YARR and RLBench branches (ie not the ones from stepjam), but I'm still running into this issue. I was wondering if you have encountered this as well, and whether you know what the fix might be?

Thanks in advance!

MohitShridhar, 2023-01-06:
@k2so3, this is a strange error. Are you sure the repos are on `peract` branches for both YARR and RLBench? Do you have any other installations of RLBench in your python env? Maybe double check on a python console if the right `rlbench` is being imported.

This error looks like it's coming from [here](https://github.com/MohitShridhar/RLBench/blob/peract/rlbench/action_modes/action_mode.py#L34). In PerAct, we have one extra dimension for `ignore_collisions` in addition to the `gripper_open` from ARM. This is probably causing the `1 != 2` mismatch error.

k2so3, 2023-01-06:
Thanks for the tips! I'll take a closer look tomorrow and report back with the results. I'm quite sure both YARR and RLBench are on the `peract` branches, but I'll double check and let you know!

k2so3, 2023-01-06:
Hi Mohit, thanks again for your help! Turns out the issue was because I followed the installation instructions from `stepjam/master` for `RLBench` and used  `pip install .`, instead of doing `python setup.py develop`. Looks like the evaluation works now without any issues!

closed 2023-01-06 by k2so3