peract/peract #10: AR controller for real data collection

geyuying, 2023-01-10:
Hi,

When you use an HTC Vive controller to collect demonstrations,  do you also use a VR headset? We currently find that the controller can not be paired without a headset in steam.

Thanks for your help!

MohitShridhar, 2023-01-11:
@geyuying, I do not own a headset. Just a controller and base station. It's possible to use the controller without the headset by following this guide: https://www.roadtovr.com/how-to-use-the-htc-vive-tracker-without-a-vive-headset/
You basically have to set `requireHmd: false`. 

For interfacing with ROS, I used this: https://github.com/uts-magic-lab/htc_vive_teleop_stuff

Hope this help!

geyuying, 2023-01-11:
Thanks a lot for your kind and timely help!

closed 2023-01-11 by geyuying