peract/peract #11: ignore_collision not part of obs_dict  variable?

ShogoAkiyama, 2023-01-16:
Hello, first of all thank you for sharing the code. amazing work!

I have a question about the `ignore_collision` variable in the replay buffer.

Currently, current_poses `ignore_collision` is inserted  in to the replay memory here
https://github.com/peract/peract/blob/main/agents/c2farm_lingunet_bc/launch_utils.py#L174

Shouldn't the agent predict the next_poses `ignore_collision` and it be inserted to obs_dict instead?

Thank you in advance!

MohitShridhar, 2023-01-17:
@ShogoAkiyama, `ignore_collision` is used to supervise the agent during training. It's a ground-truth label for activating collision avoidance, and not really part of the "input observation".

ShogoAkiyama, 2023-01-17:
Thank you for replay!
Sorry for the lack of explanation.

I tried to train a single task of `slide_block_to_target` in PerAct and it resulted in selecting a path to avoid when pressing the block as shown in the video below.
This is probably due to the fact that the data inserted into the replay memory when the block is first pressed during training has `ignore_collision=False`. (The learning loss is blue).

So I changed the code as follows to predict `ignore_collision` in `next_state `when learning.
```
final_obs = {
    ...
    'ignore_collisions': [int(obs_tm1.ignore_collisions)],
}
```
https://github.com/peract/peract/blob/main/agents/c2farm_lingunet_bc/launch_utils.py#L192-L198

This is changed `ignore_collision=True` when pressing block, which is a good result. (The learning loss is pink)

My question is, I think your code predicts ignore_collision in the `current state t`. Shouldn’t it use ignore_collision in the `next state t+1`?

Thank you.

[image]

[image]

MohitShridhar, 2023-01-18:
@ShogoAkiyama this is a good catch. Thanks for the detailed info! 

It looks like I introduced this bug while refactoring because the pre-trained models don't have this issue. I will look into this ASAP

closed 2023-02-13 by ShogoAkiyama