peract/peract #12: setup.py: parse error at "'+https:/'"

emrecanacikgoz, 2023-01-22:
I faced the issue below when I try to run `python setup.py develop` for peract part. I am using setuptools version **65.6.3**. When I include **"@"** in front of `git+https://github.com/openai/CLIP.git` in **requirements.txt**, it successfully works. What is the problem in here, any idea?

**ERROR:**
```
(peract_env) user:~/peract$ python setup.py develop
error in peract setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Parse error at "'+https:/'": Expected string_end
```

MohitShridhar, 2023-01-22:
@emrecanacikgoz, it seems like older versions of `distutils` don't support this format: https://github.com/sdispater/pendulum/issues/187

emrecanacikgoz, 2023-01-23:
Thanks for the reply @MohitShridhar. I've tried it before, exactly how it is suggested in the link and it is still giving the same error. I believe it is due to different problems; mine is caused by a "parsing" error.

emrecanacikgoz, 2023-01-23:
Since I already pip install that, I am not sure whether it is that important or not. As I mentioned, when I add "@" operator in front of that inside requirements.txt file, issue solved. However I am curious about; when I want to run eval.py for evaluate the pre-trained agent with following args:
```
cd $PERACT_ROOT
CUDA_VISIBLE_DEVICES=0 python eval.py \
    rlbench.tasks=[open_drawer] \
    rlbench.task_name='multi' \
    rlbench.demo_path=$PERACT_ROOT/data/val \
    framework.gpu=0 \
    framework.logdir=$PERACT_ROOT/ckpts/ \
    framework.start_seed=0 \
    framework.eval_envs=1 \
    framework.eval_from_eps_number=0 \
    framework.eval_episodes=10 \
    framework.csv_logging=True \
    framework.tensorboard_logging=True \
    framework.eval_type='last' \
    rlbench.headless=False
```
I am getting a numpy error: `AttributeError: module 'numpy' has no attribute 'typeDict'`. Can it be related to that, about clip setup? Below is the complete error:
```
Traceback (most recent call last):
  File "eval.py", line 23, in <module>
    from agents import c2farm_lingunet_bc
  File "/home/user/Desktop/rl-project/peract_main/peract/agents/c2farm_lingunet_bc/__init__.py", line 1, in <module>
    import agents.c2farm_lingunet_bc.launch_utils
  File "/home/user/Desktop/rl-project/peract_main/peract/agents/c2farm_lingunet_bc/launch_utils.py", line 19, in <module>
    from helpers import demo_loading_utils, utils
  File "/home/user/Desktop/rl-project/peract_main/peract/helpers/utils.py", line 2, in <module>
    import pyrender
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/pyrender/__init__.py", line 8, in <module>
    from .mesh import Mesh
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/pyrender/mesh.py", line 9, in <module>
    import trimesh
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/trimesh/__init__.py", line 15, in <module>
    from .base import Trimesh
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/trimesh/base.py", line 9, in <module>
    from . import ray
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/trimesh/ray/__init__.py", line 1, in <module>
    from .import ray_triangle
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/trimesh/ray/ray_triangle.py", line 7, in <module>
    from .ray_util import contains_points
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/trimesh/ray/ray_util.py", line 4, in <module>
    from .. import bounds
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/trimesh/bounds.py", line 5, in <module>
    from . import convex
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/trimesh/convex.py", line 21, in <module>
    from scipy import spatial
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/scipy/spatial/__init__.py", line 97, in <module>
    from .kdtree import *
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/scipy/spatial/kdtree.py", line 8, in <module>
    import scipy.sparse
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/scipy/sparse/__init__.py", line 229, in <module>
    from .base import *
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/scipy/sparse/base.py", line 8, in <module>
    from .sputils import (isdense, isscalarlike, isintlike,
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/scipy/sparse/sputils.py", line 17, in <module>
    supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/scipy/sparse/sputils.py", line 17, in <listcomp>
    supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
  File "/home/user/miniconda3/envs/peract_env/lib/python3.8/site-packages/numpy/__init__.py", line 284, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'typeDict'
```

MohitShridhar, 2023-01-24:
Hmm.. these errors seem to be specific to your env/setup. For the second error see: https://stackoverflow.com/questions/74852225/attributeerror-module-numpy-has-no-attribute-typedict

MohitShridhar, 2023-01-25:
Closing due to pip and numpy specific issues.

closed 2023-01-25 by MohitShridhar