peract/peract #13: [UTF-8 Decoding error] regarding open "variation_descriptions.pkl" provided in pre-generated data,

berniewang8177, 2023-02-06:
Hi,

Thank you for sharing your great work on github.

I'm wondering what is a correct way to open "variation_descriptions.pkl" ?
`with open("variation_descriptions.pkl") as f: pickle.load(f)` currently gives me UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 0: invalid start byte.
My device is a mac laptop.

Besides, I have checked the provided code "descs = demo._observations[0].misc['descriptions']" which get descriptions from "demo = rlbench_utils.get_stored_demos". However, in current rlbench [utils](https://github.com/stepjam/RLBench/blob/master/rlbench/utils.py)  there's no key "descriptions" added to the `mics` variable.

Could author provide some guidance here? 

Thank you very much!

MohitShridhar, 2023-02-07:
@berniewang8177 

I think you are missing `rb`. So like: 
```python
with open("variation_descriptions.pkl", 'rb') as f: 
    pickle.load(f)
```
See this for more info: https://stackoverflow.com/questions/32957708/python-pickle-error-unicodedecodeerror

> rlbench [utils](https://github.com/stepjam/RLBench/blob/master/rlbench/utils.py) there's no key "descriptions" added to the mics variable.

Because `stepjam/RLBench/master` is the wrong repo and branch. The correct repo and branch: https://github.com/MohitShridhar/RLBench/blob/peract/rlbench/utils.py#L91

berniewang8177, 2023-02-08:
Thank you very much for your information!
Problem solved!

closed 2023-02-08 by berniewang8177

Flxiang1, 2025-04-10:
> Hi,
>
> Thank you for sharing your great work on github.
>
> I'm wondering what is a correct way to open "variation_descriptions.pkl" ? `with open("variation_descriptions.pkl") as f: pickle.load(f)` [...]

Hello, I have also encountered the same problem as you:                                                                                                                 
[2025-04-10 21:15:10,361][root][INFO] - CWD:/home/sdc/flx/hdp/myperact/peract/logs/multi_18T/PERACT_BC
[2025-04-10 21:15:10,362][root][INFO] - Starting seed 0.
Process Process-6:2:
Traceback (most recent call last):
File "/home/sdc/new_envs/hdp/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
self.run()
File "/home/sdc/new_envs/hdp/lib/python3.10/multiprocessing/process.py", line 108, in run
self._target(*self._args, **self._kwargs)
File "/home/sdc/flx/hdp/myperact/peract/agents/peract_bc/launch_utils.py", line 250, in fill_replay
descs = demo._observations[0].misc['descriptions']
KeyError: 'descriptions                                                                                                                                                                         
Excuse me, how did you solve it?