peract/peract #15: Dataset generator py fails due to limsimExtROS2.so file error

vidhiJain, 2023-03-06:
I am looking into dataset_generator example shown here and it seems that the error is with CoppeliaSim not able to open `libsimExtROS2.so` or related file.
```
In ~/peract_root/RLBench/tools$ python dataset_generator.py --tasks=open_drawer                             --save_path=$PERACT_ROOT/data/val                             --image_size=128,128                             --renderer=opengl                             --episodes_per_task=10                             --processes=1                             --all_variations=True

error: library ( /home/vidhij/peract_root/CoppeliaSim_20_04/libsimExtROS2.so ) load: "Cannot load library /home/vidhij/peract_root/CoppeliaSim_20_04/libsimExtROS2.so: (libstd_srvs__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory)"
Process 0 failed collecting task open_drawer (variation: 1, example: 0). Skipping this task/variation.
The call failed on the V-REP side. Return value: -1

Process 0 finished
[CoppeliaSim:loadinfo]   done.
Data collection done!
Process 0 failed collecting task open_drawer (variation: 1, example: 0). Skipping this task/variation.
The call failed on the V-REP side. Return value: -1
```
 I am able to launch the simulator standalone using `~/peract_root/CoppeliaSim_20_04$ bash coppeliasim.sh`.  I am able to add objects and run simulation physics. The startup logs to show a similar error for ROS setup. 
 ```
 [CoppeliaSim:loadinfo]   plugin 'ROS': loading...
[simExtROS:error]   ROS master is not running
[CoppeliaSim:error]   plugin 'ROS': load failed (failed initialization).
[CoppeliaSim:loadinfo]   plugin 'ROS2': loading...
error: library ( /home/vidhij/peract_root/CoppeliaSim_20_04/libsimExtROS2.so ) load: "Cannot load library /home/vidhij/peract_root/CoppeliaSim_20_04/libsimExtROS2.so: (libstd_srvs__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory)"
[CoppeliaSim:error]   plugin 'ROS2': load failed (could not load). The plugin probably couldn't load dependency libraries. For additional infos, modify the script 'libLoadErrorCheck.sh', run it and inspect the output.
[CoppeliaSim:loadinfo]   plugin 'RRS1': loading...
[CoppeliaSim:loadinfo]   plugin 'RRS1': load succeeded.
``` 
Since we don't need ROS for data generation, is there some way to try-catch this error and use the simulator to generate the data?

MohitShridhar, 2023-03-06:
@vidhiJain, this is indeed strange. I have never installed ROS2 for CoppeliaSim, but maybe this issue comes up with 20.04. 

You could try an older version of CoppeliaSim, or look into installing ROS2 extensions:
https://github.com/CoppeliaRobotics/simExtROS2
https://forum.coppeliarobotics.com/viewtopic.php?t=8692

vidhiJain, 2023-03-13:
I am trying to run the older version of CoppeliaSim 18.04 inside docker, but not able to get PyRep working. Here is the issue for more details:
`https://github.com/CoppeliaRobotics/docker-image-coppeliasim/issues/4#issue-1622122047`
@MohitShridhar Is there a docker image that worked for you? Could you specify your system configurations and I'll try to replicate them?

MohitShridhar, 2023-03-14:
@vidhiJain, emailed you regarding the Docker image. Sorry i haven't had time to look into this; I have been swamped with thesis work.

vidhiJain, 2023-03-15:
Thanks, @MohitShridhar! I found the solution finally on Ubuntu 18.04 and 20.04 with python3.8. 
Here are the main points I had to ensure:
- coppeliaSim 4.1 version is downloaded for the correct ubuntu version. Otherwise, PyRep will not work. 
- If the PyRep build fails, especially with `pyconfig.h` file missing, 
      run `sudo apt-get install python3.8-dev` and again try `pip install .` in PyRep directory.

closed 2023-03-15 by vidhiJain