peract/peract #16: Question about the error [free(): invalid pointer] during Evaluation

sxy7147, 2023-03-14:
Hi, authors

Thanks for sharing your code! The code is very helpful and I have successfully re-trained and evaluated your networks. 
Currently, I'm trying to modify the networks and the observation settings. I have successfully trained a new network based on new observation settings, but it reports an error when I **evaluate the new network.** The error is 

> free(): invalid pointer

The error occurs when it tries to start the `env_runner`, before the `agent` starts to `act`.

I have no idea how to solve this problem. I think maybe the reason is that I have changed the observation settings. During training, I modified the` agent/launch_utils.py` to change the observation, but I don't know where to change the observation during inference.

Do you have any idea about how to solve this error? 

Thanks in advance!

sxy7147, 2023-03-15:
It just prints some information and then reports the error, sorry I don't know how to find the full stacktrace.
The print information is:

```
[2023-03-15 13:10:38,130][root][INFO] -                                                                                            
enabled: false                                                                                                                     
camera_resolution:                                                                                                                 
- 1280                                                                                                                             
- 720                                                                                                                              
fps: 30                                                                                                                            
rotate_speed: 0.005                                                                                                                
save_path: /mnt/data2/zy/PerAct/logs/Affordance_031202_withAFF_allView_AffordanceAndActor/multi/Affordance/seed0/results_7000      
                                                                                                                                   
[2023-03-15 13:10:38,285][root][INFO] - Using env device cuda:0.                                                                   
[2023-03-15 13:10:38,910][root][INFO] - Evaluating seed 0.    

free(): invalid pointer
```

MohitShridhar, 2023-03-15:
@sxy7147, it's hard to say what's happening here. Can you use `gdb` or something to figure out where this memory issue is coming from?

Also, maybe briefly explain your changes to `launch_utils.py`?

sxy7147, 2023-03-20:
After debugging, I find that it is the import of `open3d` that leads to the error `free(): invalid pointer.` After removing the `open3d`, everything works well.
I still have no idea why I can't use open3d during evaluation, but I have fixed it anyway.

MohitShridhar, 2023-03-20:
@sxy7147, sorry where is `open3d` used? Is this something you added?

sxy7147, 2023-03-21:
Yeah, I use `open3d` in the `agent.update()` (for training) and `agent.act()` (for inference) to visualize the pointclouds and some additional heatmaps. 
During training, everything works well. But during inference, the import of `open3d` will lead to the error.
Now I have removed `open3d` and saved the points as `npy` for visualization later, and now I can successfully run my code. If we all have no idea about this strange error, I think I can settle this problem in this way. Thank you very much!

MohitShridhar, 2023-03-22:
Glad you fixed it!

closed 2023-03-22 by MohitShridhar