peract/peract #17: Repetitive keypoints added in replay buffer

qianLyu, 2023-03-14:
Hi @MohitShridhar, thanks for sharing your amazing work! I have a question about repetitive (maybe?) keypoints added in the replay buffer. The related code is the loop [here](https://github.com/peract/peract/blob/6e92e64a72e27a12835204ece48ed30ad8fdd74f/agents/peract_bc/launch_utils.py#L165)

From my understanding, the func "_add_keypoints_to_replay()" will be called every 10 steps if "demo_augmentation" is enabled, so that the loop mentioned above will also be revisited many times in a single demo. Thus, there will be a lot of repetitive "obs_tp" data added to the replay buffer since they are generated by the same keypoint pairs.

If this is the case, I'm wondering if those repetitive keypoints data would be beneficial to our training process? Thank you!

MohitShridhar, 2023-03-14:
@qianLyu, related to this? https://github.com/peract/peract/issues/6#issuecomment-1355555980

qianLyu, 2023-03-15:
Yes, that's exactly the same issue. Thanks for your reply!

closed 2023-03-15 by qianLyu