peract/peract #19: Question for efficient replay pickle file handling

sxy7147, 2023-03-30:
Hello Authors,

I have noticed that saving the replay pickle files before training is a time-consuming process. It requires me to repeatedly save the same replay pickle files for each experiment, which can be quite cumbersome. Therefore, I am curious if there is a way to directly load in the saved pickle files, which could save me a lot of time.

Thank you in advance!

MohitShridhar, 2023-03-30:
@sxy7147, this is true. Saving pickle files every time is slow. But this also depends on your application. Maybe the pre-processing step is different for each run, and if so, you might have to regenerate the pickle files anyway.

You could save pickle files with a hash, and next time, check if that hash already exists, so you can avoid re-saving the same file.

MohitShridhar, 2023-05-06:
@sxy7147, checkout this: https://github.com/peract/peract#fire-hotfix

sxy7147, 2023-05-06:
Your modification has been tremendously helpful to me. Thank you very much!