peract/peract #21: How to train with real-world dataset.

maineJiang, 2023-05-11:
Your work is appealing, thanks a lot for the effort. I am currently trying to train peract with real world task. I have found your explanation about data collection is sufficient. However, how to train a model in real world remains unclear. For example, using the real world dataset you provided, the first problem I counter is that the RLbench is asking for 4 cameras' image to form a demo. Do you have any advice on how we can modify the training code to support one camera real world training? Thanks again.

MohitShridhar, 2023-05-11:
@maineJiang, did you run `train.py` with `rlbench.cameras=['front']`? 

If it's still throwing errors:
1. You might have to comment-out [this line](https://github.com/MohitShridhar/RLBench/blob/peract/rlbench/utils.py#L116) that checks for multi-camera images.
2. `DEPTH_SCALE` is also different for real-world cameras.

maineJiang, 2023-05-15:
Your suggestion is accurate. I appreciate your in time response. Now the training process seems good. 
Also a reminder for others, if you encounter the problem of could not create GL context, you may try to comment the 2 lines of collecting summary.

MohitShridhar, 2023-05-15:
Nice!

closed 2023-05-15 by MohitShridhar

TongZhangTHU, 2023-05-25:
> @maineJiang, did you run `train.py` with `rlbench.cameras=['front']`?
>
> If it's still throwing errors:
>
> 1. You might have to comment-out [this [...]

@MohitShridhar Hi, if I follow your codes for real-robot experiments, what should `DEPTH_SCALE` be? I can't find anywhere the `DEPTH_SCALE` is specified in the codes for real-robot experiments.

Another question is, if the images in my data are`1280*720` and I think it will be resized to 128*128 [here](https://github.com/MohitShridhar/RLBench/blob/peract/rlbench/utils.py#L205), then do I need to make any change to the camera intrinsics?

Thank you so much for your kind help and looking forward to your reply!

maineJiang, 2023-05-25:
@TongZhangTHU DEPTH_SCALE aims at converting your depth image to meter. Like if your depth is in mm, you can set DEPTH_SCALE = 1000 to convert it to meter. About the resized image, I believe you need to give a updated camera intrinsics(actually just scale the fx fy cx cy with the resized ratio). You can use open3d to determine if your resized intrainsics is right.