peract/peract #22: Colab Install Dependencies Error

SageRobots, 2023-05-19:
The first cell in the Colab notebook has an error when I try to run it. Any ideas on how to fix this?

Looking in indexes: https://pypi.org/simple, https://us-python.pkg.dev/colab-wheels/public/simple/
Collecting git+https://github.com/openai/CLIP.git
  Cloning https://github.com/openai/CLIP.git to /tmp/pip-req-build-_wi6f7w4
  Running command git clone --filter=blob:none --quiet https://github.com/openai/CLIP.git /tmp/pip-req-build-_wi6f7w4
  Resolved https://github.com/openai/CLIP.git to commit a9b1bf5920416aaeaec965c25dd9e8f98c864f16
  Preparing metadata (setup.py) ... done
Collecting scipy==1.4.1
  Downloading scipy-1.4.1.tar.gz (24.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 24.6/24.6 MB 15.8 MB/s eta 0:00:00
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Installing build dependencies ... error
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

MohitShridhar, 2023-05-20:
@SageRobots, I just changed `scipy==1.4.1` and `torch==1.7.1` to `scipy` and `torch`. Checkout the updated [Colab notebook](https://colab.research.google.com/drive/1HAqemP4cE81SQ6QO1-N85j5bF4C0qLs0?usp=sharing).

Hope this helps!

MohitShridhar, 2023-05-23:
Should be fixed!

closed 2023-05-23 by MohitShridhar