peract/peract #24: Size mismatch error when batch size > 1

SageRobots, 2023-06-06:
Everything seems to work when I have batch_size: 1 in the config. When I switch to batch_size: 2 train.py runs fine, but eval.py results in this error.

  File "/home/user/School/peract/agents/peract_bc/qattention_peract_bc_agent.py", line 666, in load_weights
    self._q.load_state_dict(merged_state_dict)
  File "/home/user/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1671, in load_state_dict
    raise RuntimeError('Error(s) in loading state_dict for {}:\n\t{}'.format(
RuntimeError: Error(s) in loading state_dict for QFunction:
        size mismatch for _voxelizer._ones_max_coords: copying a param with shape torch.Size([2, 65536, 1]) from checkpoint, the shape in current model is torch.Size([1, 65536, 1]).
        size mismatch for _voxelizer._flat_output: copying a param with shape torch.Size([14856912]) from checkpoint, the shape in current model is torch.Size([7428456]).
        size mismatch for _voxelizer._tiled_batch_indices: copying a param with shape torch.Size([2, 65536, 1]) from checkpoint, the shape in current model is torch.Size([1, 65536, 1]).
        size mismatch for _voxelizer._index_grid: copying a param with shape torch.Size([2, 102, 102, 102, 3]) from checkpoint, the shape in current model is torch.Size([1, 102, 102, 102, 3]).

SageRobots, 2023-06-06:
I did a hack fix in qattention_peract_bc_agent.py 

`    def load_weights(self, savedir: str):
        device = self._device if not self._training else torch.device('cuda:%d' % self._device)
        weight_file = os.path.join(savedir, '%s.pt' % self._name)
        state_dict = torch.load(weight_file, map_location=device)

        # load only keys that are in the current model
        merged_state_dict = self._q.state_dict()
        for k, v in state_dict.items():
            if not self._training:
                k = k.replace('_qnet.module', '_qnet')
            if k in merged_state_dict:
                merged_state_dict[k] = v
            else:
                if '_voxelizer' not in k:
                    logging.warning("key %s not found in checkpoint" % k)
        **if not self._training:
            # reshape voxelizer weights
            merged_state_dict['_voxelizer._ones_max_coords'] = merged_state_dict['_voxelizer._ones_max_coords'][0:1]
            flat_shape = merged_state_dict['_voxelizer._flat_output'].shape[0]
            merged_state_dict['_voxelizer._flat_output'] = merged_state_dict['_voxelizer._flat_output'][0:flat_shape // 2]
            merged_state_dict['_voxelizer._tiled_batch_indices'] = merged_state_dict['_voxelizer._tiled_batch_indices'][0:1]
            merged_state_dict['_voxelizer._index_grid'] = merged_state_dict['_voxelizer._index_grid'][0:1]**
        self._q.load_state_dict(merged_state_dict)
        print("loaded weights from %s" % weight_file)`

baldeeb, 2023-08-18:
The buffers in the `VoxelGrid` do not need to be saved. The `config.yaml` is sufficient to re-create the buffered parameters for a saved model. To avoid saving the buffers to the state_dict, you could pass `persistent=False` to the `register_buffer` functions in the `VoxelGrid`. This way the buffers will not be saved and instead re-created.

akushonkamen, 2026-10-17:
Hi, I'd like to take this one.

**Root cause:** `VoxelGrid.__init__` (`voxel/voxel_grid.py:40-93`) registers all 11 buffers with the default `persistent=True`, and several of them (`_ones_max_coords`, `_flat_output`, `_tiled_batch_indices`, `_index_grid`) are shaped by `batch_size`. `save_weights` therefore persists batch-size-dependent buffers, and loading such a checkpoint into a model built with a different batch size fails in `load_weights` (`agents/peract_bc/qattention_peract_bc_agent.py:666`) with the reported `size mismatch`.

**Plan:** mark the VoxelGrid buffers `persistent=False` (as @baldeeb suggested above) so they are reconstructed from constructor args and checkpoints become batch-size-agnostic. Both agents' `load_weights` already skip checkpoint-only `\_voxelizer.\*` keys (`agents/peract_bc/qattention_peract_bc_agent.py:661-665`, `agents/c2farm_lingunet_bc/qattention_lingunet_bc_agent.py:665`), so existing checkpoints remain loadable. I'll include a regression test.

*(Disclosure: this comment and the upcoming fix were drafted with AI assistance and reviewed by a human.)*

akushonkamen, 2026-10-17:
Opened PR #69 with a fix: all 12 `VoxelGrid` buffers are now registered with `persistent=False` so checkpoints no longer store the batch-size-dependent buffers (`_ones_max_coords`, `_flat_output`, `_tiled_batch_indices`, `_index_grid`), and old checkpoints keep loading since both agents' `load_weights` already skip `_voxelizer.*` keys. Includes a regression test reproducing the batch 2 → 1 load failure. Details and test evidence in the PR.