peract/peract #26: Questions in reset_to_demo and rollout_generator

TongZhangTHU, 2023-08-18:
Hi Mohit,

Thank you for your exceptional implementation! I would like to ask two questions regarding specific aspects of your code.

I have noticed that even after calling the [reset_to_demo](https://github.com/peract/peract/blob/main/helpers/custom_rlbench_env.py#L180) function, the initial object pose configurations sometimes (although not always) remain different compared to the data. In these situations, I observe that the [init_episode](https://github.com/MohitShridhar/RLBench/blob/peract/rlbench/backend/scene.py#L128) function attempts different numbers of times. I hypothesize that the sampling-based planner used in RLBench introduces a degree of randomization, which leads to differnet outcomes in the [task validation](https://github.com/MohitShridhar/RLBench/blob/peract/rlbench/backend/scene.py#L136) process. Have you made similar observations? If so, do you have any suggestions on how this issue might be resolved?

Is there a bug [here](https://github.com/MohitShridhar/YARR/blob/peract/yarr/utils/rollout_generator.py#L81)? It seems that changing the condition to if record_enabled and (transition.terminal or timeout or step == episode_length - 1):  might be more reasonable, as in Python the 'and' operator has a higher precedence than the 'or' operator. Could you please confirm?

Thank you in advance for your time and assistance in addressing these issues.

MohitShridhar, 2023-08-19:
@TongZhangTHU 

1. Yeah, this could be possible. I didn't have time to manually verify that `reset_to_demo` works for every variation of every task. Which tasks specifically have this issue? It might be easier to fix them individually.
2. Good catch! You are right

TongZhangTHU, 2023-08-19:
@MohitShridhar 

While testing on the 'open_fridge' task, I encountered this issue.

MohitShridhar, 2023-08-21:
Sorry I don't have much experience with `open_fridge`, but it might be worth fixing that task in particular. Something could be wrong with `ttm` file or `open_fridge.py`.  I am not sure where the non-determinism is coming from. Maybe it's worth asking on the RLBench github.

TongZhangTHU, 2023-08-22:
Thank you for your suggestion!

closed 2023-08-22 by TongZhangTHU