peract/peract #30: An error occurred while running eval.py

zqy-an, 2023-09-05:
Hello!
Your work looks great!

When I run the eval.py file with reference to readme.md, I get an error:
```
"/home/zqy/code/robotics/RVT/rvt/libs/YARR/yarr/utils/rollout_generator.py", line 40, in generator
    obs = env.reset_to_demo(eval_demo_seed)
  File "/home/zqy/code/robotics/peract/helpers/custom_rlbench_env.py", line 188, in reset_to_demo
    self._task.set_variation(d.variation_number)
AttributeError: 'Demo' object has no attribute 'variation_number'
```
I find the get_stored_demos function in rlbench/utils.py, and it doesn't define  'variation_number'

MohitShridhar, 2023-09-05:
@zqy-an, are you using my fork of RLBench: https://github.com/MohitShridhar/RLBench/tree/peract ?

You can check by:
```
import rlbench
print(rlbench.__path__)
```

zqy-an, 2023-09-09:
`['/home/xxx/code/robotics/RLBench/rlbench']`
I'm sure RLBench was used. 
I think the problem is with the get_stored_demos function in rlbench/utils.py
[image]

MohitShridhar, 2023-09-12:
@zqy-an, my guess is the wrong RLBench is being used when running `data_generator.py`. You can manually open the pickle files and check if the variation number exists.

MohitShridhar, 2024-04-24:
Closing due to inactivity.

closed 2024-04-24 by MohitShridhar