peract/peract #31: The `low_dim_state` in RLBench.

AoqunJin, 2023-10-14:
I found that `low_dim_state` is used in PerAct. How much impact will this have on the experimental results?

In `peract/agents/peract_bc/launch_utils.py`

QAttentionPerActBCAgent `include_low_dim_state=True`.

MohitShridhar, 2023-10-15:
@AoqunJin, the `gripper_open` state and `time_left` are probably important for the agent to figure out which part of the trajectory it's at, e.g., since the gripper is closed, the object has probably been grasped. But I don't think I ran ablations. It might be good to investigate this.

AoqunJin, 2023-10-16:
Thanks for your patient reply, I will try to test them in my experiments and ablation experiments, based on PerAct. 😊😊😊

closed 2023-10-16 by AoqunJin

AoqunJin, 2023-10-18:
[image]