peract/peract #35: Questions about action space

SatireY, 2023-12-05:
Thank you for your outstanding work!
I'm also running my experiments in an RLBench environment, and I have a question for you: Do you know what the three-dimensional position range of the grriper is?
I looked up the RLBench source code and did not find the relevant instructions, looking forward to your reply, thank you.

MohitShridhar, 2023-12-05:
@SatireY, the range is here: https://github.com/peract/peract/blob/main/conf/config.yaml#L14

`scene_bounds: [-0.3, -0.5, 0.6, 0.7, 0.5, 1.6]  # [x_min, y_min, z_min, x_max, y_max, z_max]` in meters

SatireY, 2023-12-09:
Oh! Thanks for the quick reply, wanted to check with you if this range is defined by the RLBench environment? Where did you find it?

SatireY, 2023-12-09:
Oh! Thanks for the quick reply, wanted to check with you if this range is defined by the RLBench environment? Is the range consistent from task to task? Where did you find it?

SatireY, 2023-12-09:
> @SatireY, the range is here: https://github.com/peract/peract/blob/main/conf/config.yaml#L14
> 
> `scene_bounds: [-0.3, -0.5, 0.6, 0.7, 0.5, 1.6] # [x_min, y_min, z_min, x_max, y_max, z_max]` in meters

I wrote a script to try to find the range of actions in the collected data, and I got results that differed from scene_bounds. Here is my script and the results.
```
# find the range of gripper_posed
import os
import pickle
import numpy as np

def extract_gripper_pose_data(directory):
    print("Extracting gripper pose data from directory: ", directory)
    all_data = []

    for root, dirs, files in os.walk(directory):
        if "low_dim_obs.pkl" in files:
            file_path = os.path.join(root, "low_dim_obs.pkl")
            with open(file_path, "rb") as file:
                data = pickle.load(file)
                for item in data:
                    gripper_pose = getattr(item, "gripper_pose")
                    # print(type(gripper_pose))
                    all_data.append(gripper_pose)
    # print(len(all_data))
    # print(all_data)
    if not all_data:
        return None

    all_data_np = np.array(all_data)

    # print(all_data_np)

    max_values = np.max(all_data_np, axis=0)
    min_values = np.min(all_data_np, axis=0)
    print("read gripper pose data successfully! ✔️")
    return all_data_np, max_values, min_values

directory = "dataroot/"
all_data_np, max_values, min_values = extract_gripper_pose_data(directory)
print("Max values:", max_values)
print("Min values:", min_values)
```
[image]
How can this be explained?

MohitShridhar, 2024-01-13:
@SatireY, some gripper poses in the trajectory might be outside these bounds because of weird RRT motion paths. Only the gripper poses of the keypoints matter. You can try computing the bounds of just keypoint actions.

closed 2024-04-22 by MohitShridhar