peract/peract #41: Error when install YARR

kobehon, 2024-03-21:
`python setup.py develop
running develop
/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
running egg_info
writing yarr.egg-info/PKG-INFO
writing dependency_links to yarr.egg-info/dependency_links.txt
writing requirements to yarr.egg-info/requires.txt
writing top-level names to yarr.egg-info/top_level.txt
reading manifest file 'yarr.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'yarr.egg-info/SOURCES.txt'
running build_ext
Creating /home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/yarr.egg-link (link to .)
yarr 0.1 is already the active version in easy-install.pth

Installed /home/lenovo/peract3/PyRep/YARR
Processing dependencies for yarr==0.1
Traceback (most recent call last):
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3070, in _dep_map
    return self.__dep_map
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2863, in __getattr__
    raise AttributeError(attr)
AttributeError: _DistInfoDistribution__dep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/requirements.py", line 35, in __init__
    parsed = _parse_requirement(requirement_string)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/_parser.py", line 64, in parse_requirement
    return _parse_requirement(Tokenizer(source, rules=DEFAULT_RULES))
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/_parser.py", line 82, in _parse_requirement
    url, specifier, marker = _parse_requirement_details(tokenizer)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/_parser.py", line 120, in _parse_requirement_details
    specifier = _parse_specifier(tokenizer)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/_parser.py", line 216, in _parse_specifier
    parsed_specifiers = _parse_version_many(tokenizer)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/_parser.py", line 231, in _parse_version_many
    tokenizer.raise_syntax_error(
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/_tokenizer.py", line 165, in raise_syntax_error
    raise ParserSyntaxError(
pkg_resources.extern.packaging._tokenizer.ParserSyntaxError: .* suffix can only be used with `==` or `!=` operators
    PyYAML (>=5.1.*)
            ~~~~~~^

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "setup.py", line 30, in <module>
    setuptools.setup(
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/__init__.py", line 104, in setup
    return distutils.core.setup(**attrs)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/dist.py", line 967, in run_command
    super().run_command(command)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/command/develop.py", line 34, in run
    self.install_for_development()
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/command/develop.py", line 126, in install_for_development
    self.process_distribution(None, self.dist, not self.no_deps)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/setuptools/command/easy_install.py", line 785, in process_distribution
    distros = WorkingSet([]).resolve(
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/__init__.py", line 803, in resolve
    new_requirements = dist.requires(req.extras)[::-1]
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2786, in requires
    dm = self._dep_map
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3072, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3082, in _compute_dependencies
    reqs.extend(parse_requirements(req))
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3135, in __init__
    super().__init__(requirement_string)
  File "/home/lenovo/anaconda3/envs/peract4/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/requirements.py", line 37, in __init__
    raise InvalidRequirement(str(e)) from e
pkg_resources.extern.packaging.requirements.InvalidRequirement: .* suffix can only be used with `==` or `!=` operators
    PyYAML (>=5.1.*)
            ~~~~~~^`

MohitShridhar, 2024-03-21:
@kobehon, try installing pyyaml manually: `pip install pyyaml==5.1`

kobehon, 2024-03-22:
> @kobehon, try installing pyyaml manually: `pip install pyyaml==5.1`

I originally have PyYAML-6.0.1. If I run this command, it uninstalls 6.0.1 and installs 5.1. 

However, after trying this, it still doesn't work as shown in the pictures below. 

I am not sure whether you have seen the error: AttributeError: _DistInfoDistribution__dep_map    ( shown also in the pictures)

Please advise. Thank. 

[image]

[image]

[image]

tanmayyb, 2024-05-17:
Omegaconf package's `METADATA` file is causing this error when a description generator is used:
```
pkg_resources.extern.packaging.requirements.InvalidRequirement: .* suffix can only be used with `==` or `!=` operators
    PyYAML (>=5.1.*)
```
They stoped maintaining omegaconf after version 2.0.6 and this PEP-related bug persists till today. I'm sure why the original authors of PerAct did not encounter this issue. Anyways, luckily there is an easy workaround.

I am using a miniconda environment and I bypassed this issue by just changing 1 line in the `METADATA` file of the downloaded omegaconf site-package. But this fix should work for a conda or global python environment too - you just have to locate the site-packages dir being used by your commands.

Navigate to package site:
```
cd ~/miniconda3/envs/<NAME-OF-YOUR-PERACT-CONDA-ENVIRONMENT>/lib/python3.8/site-packages/omegaconf-2.0.6.dist-info
```
and edit the `METADATA` file and change:
```
Requires-Dist: PyYAML (>=5.1.*)
to
Requires-Dist: PyYAML (>=5.1)
```

Best of luck!