peract/peract #47: Error when running evaluation

LightingTom, 2024-07-22:
I tried to run the evaluation using the pretrained checkpoint [PerAct - 512 Latents](https://github.com/peract/peract/releases/download/v1.0.0/peract_600k_512latents.zip) and the val dataset provided. However, the following error occurred. 

[image]

[image]

MohitShridhar, 2024-07-22:
@LightingTom, this looks like an RLBench display issue. Is the `LD_LIBRARY_PATH` and `COPPELIASIM_ROOT` path set correctly? And running `glxgears` works without any issues?

LightingTom, 2024-07-22:
@MohitShridhar , thanks for your reply. I believe I have set the `LD_LIBRARY_PATH` and `COPPELIASIM_ROOT` correctly as described in the README file. What do you mean by running `glxgears`?

MohitShridhar, 2024-07-23:
@LightingTom, can you try `sudo apt-get install mesa-utils` and then `DISPLAY=:0 glxgears`. Does this work?

LightingTom, 2024-07-23:
I tried to run `sudo apt-get install mesa-utils` and use `export DISPLAY=:0 glxgears`. But still the same error occured
[image]

LightingTom, 2024-07-25:
I tried to run `glxgears`. It has the following bug report.

[image]

MohitShridhar, 2024-07-25:
@LightingTom this is a display issue. Checkout issues on how to fix this:
https://unix.stackexchange.com/questions/85782/error-no-protocol-specified-when-running-from-remote-machine-via-ssh

rebecajohn, 2024-09-07:
Command used : bash online_evaluation_rlbench/eval_peract.sh

Loading model from train_logs/diffuser_actor_peract.pth
Gripper workspace
Gripper workspace size: [0.75823578 1.07414986 0.79873248]
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/teamspace/studios/this_studio/PyRep/CoppeliaSim_Edu_V4_1_0_Ubuntu20_04" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb.

/commands/python: line 37: 40081 Aborted (core dumped) "$@"
/home/zeus/miniconda3/envs/cloudspace/lib/python3.10/site-packages/torchdata/datapipes/init.py:18: UserWarning:
################################################################################
WARNING!
The 'datapipes', 'dataloader2' modules are deprecated and will be removed in a
future torchdata release! Please see https://github.com/pytorch/data/issues/1196
to learn more and leave feedback.
################################################################################

deprecation_warning()