peract/peract #48: About checkpoint parameter

TengBoYuu, 2024-07-30:
Hi，thank you for your great work☺️
I download the checkpoint with 2048 latents，but I notice the shape of the parameter ‘_qnet.module.pos_encoding‘ is `torch.Size([1, 20, 20, 20, 128])`，which means `pos_encoding_with_lang = False`，did I misunderstand or there was no language condition for the checkpoint training ?
[image]

MohitShridhar, 2024-07-30:
@TengBoYuu, there is [language conditioning](https://github.com/peract/peract/blob/main/agents/peract_bc/perceiver_lang_io.py#L378), but there is no positional embeddings for those lang tokens. So "put the marker in the drawer" is treated as a bag of words.

TengBoYuu, 2024-07-31:
Okay, thank you. I see what you mean.

closed 2024-08-01 by MohitShridhar