peract/peract #50: problem while running eval script

rebecajohn, 2024-09-07:
Command : bash online_evaluation_rlbench/eval_peract.sh

output: 
Loading model from train_logs/diffuser_actor_peract.pth Gripper workspace Gripper workspace size: [0.75823578 1.07414986 0.79873248] qt.qpa.xcb: could not connect to display qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/teamspace/studios/this_studio/PyRep/CoppeliaSim_Edu_V4_1_0_Ubuntu20_04" even though it was found. This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem. Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb. /commands/python: line 37: 127766 Aborted                 (core dumped) "$@"

MohitShridhar, 2024-09-10:
@rebecajohn, looks like a display issue.

If you `sudo apt-get install mesa-utils` and run `DISPLAY=:0 glxgears`, does that work?

return-sleep, 2024-09-11:
same problem
`Xvfb :1 -screen 0 1024x768x24 & export DISPLAY=:1
glxgears` works

But when I run eval.py, something wrong happened:
  File "/home/xuh0e/anaconda3/envs/RLbench/lib/python3.8/site-packages/pyglet-2.1.dev5-py3.8.egg/pyglet/gl/xlib.py", line 117, in create_context
    return XlibContext(self, share)
  File "/home/xuh0e/anaconda3/envs/RLbench/lib/python3.8/site-packages/pyglet-2.1.dev5-py3.8.egg/pyglet/gl/xlib.py", line 152, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context

MohitShridhar, 2024-09-12:
Have you seen https://github.com/mmatl/pyrender/issues/13
and related issues?

rebecajohn, 2024-09-12:
Hi Mohit , could you plan for a docker container in a headless machine and send the docker container tag , so that i pull and do the eval in it . Thanks.

MohitShridhar, 2024-09-14:
@rebecajohn, I never got around to it sorry, but somebody was trying here: https://hub.docker.com/r/shulin16/sampleperact

rebecajohn, 2024-09-14:
Hi , its showing interactive window
[image]

closed 2024-09-15 by rebecajohn