peract/peract #64: Why are there 3 channels for the depth images?

Undefinedefity, 2025-03-12:
Why are there 3 channels for the depth images? The 1st and 2nd channels seem to be meaningless, only the 3rd channel has the real depth value. Or is there any other meaning? 

e.g.: 
```
rgb_path = "./raw/train/close_jar/all_variations/episodes/episode0/front_rgb/0.png"
depth_path = "./raw/train/close_jar/all_variations/episodes/episode0/front_depth/0.png"
```
and visualize them:

[image]

MohitShridhar, 2025-03-12:
> (peract/peract#64) <https://github.com/peract/peract/issues/64>

If I remember correctly, they are actually 3D points (x y z), from an
orthographic depth camera

Undefinedefity, 2025-03-12:
@MohitShridhar  Thanks for replying. At the beginning, I thought the same as you. But I found that this is not the 3D points (x, y, z) as I plotted them in a 3D visualizer.

[image]

MohitShridhar, 2025-03-12:
If you open front_depth/0.png, can you see anything in the image itself?
Maybe compute the mean, max, min of the values? It could be a points with
some sort of normalization.

Undefinedefity, 2025-07-17:
@MohitShridhar Hi, I came across to kown the answer (though I should come to RLbench github repo eariler). This depth image is encoded as a 24-bit RGB image, as in [https://github.com/stepjam/RLBench/blob/master/rlbench/backend/utils.py](https://github.com/stepjam/RLBench/blob/master/rlbench/backend/utils.py). In order to get the most preciese depth value (up to 1/256mm), encoding a depth value in 1 meter unit as a there-channel value in 1/256mm unit, in a way like 256-decimal:

```
channel_1/1st_value = (f / 256) / 256 
channel_2/2nd_value = (f / 256) % 256
channel_3/3rd_value = f % 256
```

The original code:

```python
DEFAULT_RGB_SCALE_FACTOR = 256000.0

def float_array_to_rgb_image(float_array,
                             scale_factor=DEFAULT_RGB_SCALE_FACTOR,
                             drop_blue=False):
  """Convert a floating point array of values to an RGB image.

  Convert floating point values to a fixed point representation where
  the RGB bytes represent a 24-bit integer.
  R is the high order byte.
  B is the low order byte.
  The precision of the depth image is 1/256 mm.

  Floating point values are scaled so that the integer values cover
  the representable range of depths.

  This image representation should only use lossless compression.

  Args:
    float_array: Input array of floating point depth values in meters.
    scale_factor: Scale value applied to all float values.
    drop_blue: Zero out the blue channel to improve compression, results in 1mm
      precision depth values.

  Returns:
    24-bit RGB PIL Image object representing depth values.
  """
  # Scale the floating point array.
  scaled_array = np.floor(float_array * scale_factor + 0.5)
  # Convert the array to integer type and clip to representable range.
  min_inttype = 0
  max_inttype = 2**24 - 1
  scaled_array = ClipFloatValues(scaled_array, min_inttype, max_inttype)
  int_array = scaled_array.astype(np.uint32)
  # Calculate:
  #   r = (f / 256) / 256  high byte
  #   g = (f / 256) % 256  middle byte
  #   b = f % 256          low byte
  rg = np.divide(int_array, 256)
  r = np.divide(rg, 256)
  g = np.mod(rg, 256)
  image_shape = int_array.shape
  rgb_array = np.zeros((image_shape[0], image_shape[1], 3), dtype=np.uint8)
  rgb_array[..., 0] = r
  rgb_array[..., 1] = g
  if not drop_blue:
    # Calculate the blue channel and add it to the array.
    b = np.mod(int_array, 256)
    rgb_array[..., 2] = b
  image_mode = 'RGB'
  image = Image.fromarray(rgb_array, mode=image_mode)
  return image
```

closed 2025-07-17 by Undefinedefity